ibnuda/dactyl-keyboard #2: How do you view the left hand side of the case?

blakedietz, 2020-04-09:
Checking out the refaktor branch and puling latest. After installing everything and generating the file I only see the right hand side of the case that's generated by the sweet web gui you created. How do I view the left hand of the board? I'm a newbie to open scad, so I'm wondering if I'm missing anything.

razerbann, 2020-04-09:
There are two options:
1- You generate the STL out of the myfile.scad generated and use any tool for 3D modeling (Fuse 360 or Tinkercad for instance) and mirror the object.
2- Or you could Add this code to the end of manuform.clj file to generate the left half and the plate to things directory
```clojure
spit "things/left-plate.scad"
      (write-scad (mirror [-1 0 0] (right-plate c)))

spit "things/left.scad"
        (write-scad (mirror [-1 0 0] (model-right c)))
```

ibnuda, 2020-04-09:
what has @razerbann said is correct. and to make it less cumbersomeyfor user, I will add an option to pick either side.

ibnuda, 2020-04-09:
fixed and deployed.

closed 2020-04-09 by ibnuda

blakedietz, 2020-04-10:
Hey, thank you for the updates. Seriously.

commit 3011c245 (2020-04-09): fixes #2.