ibnuda/dactyl-keyboard #3: Screw inserts and plates for Dactyl Lighcycle / Support for external holder

razerbann, 2020-04-09:
Hello!

I really like what you're doing here and I was wondering if you were planning on adding the screw inserts and plates for the Dactyl lightcycle variant ?
Also, would you be interested in adding support for the external holder by loligaggers (https://geekhack.org/index.php?topic=88576.100)?

ibnuda, 2020-04-09:
thanks.

for screw holes and plates for lightcycle, it should be simple. as for the external holder, i will see what i can do.

i will make it as a milestone.

razerbann, 2020-04-09:
You're very welcome.
I can help with that too, I'm familiar with the code for Dactyl and Dactyl Manuform ;)

ibnuda, 2020-04-11:
sorry for the late reply.

sure, you can start from anywhere. i will try to accommodate it.

closed 2020-04-18 by razerbann