ibnuda/dactyl-keyboard #5: Lightcycle: an object is blocking one of the thumb keys

razerbann, 2020-04-15:
Using the default configuration from the UI, I found that an object is blocking one of the thumb keys.
The following images describe the issue:
[image]
[image]
[image]

I think that it's the corner of the last key in the grid.
When I choose **Thumb Position in X direction (left/right): Farther to body**, I can still see the corner but it's no longer blocking the key.
When I choose **Thumb Position in Z direction (up/down): Higher to body**, I can no longer see the corner.

ibnuda, 2020-04-15:
ah right. I haven't checked every option exhaustively. for this problem, I think the better choice is to set the minimum thumb height. what do you think?

ibnuda, 2020-04-15:
uses lastrow:

[image]

doesn't use lastrow:
[image]

i will push the fix in the next commit.

razerbann, 2020-04-15:
> ah right. I haven't checked every option exhaustively. for this problem, I think the better choice is to set the minimum thumb height. what do you think?
> 
> edit: actually, it's the `connectors`. especially on the last row and the second column. i will fix it.

Sorry for the delay of my reply.
Yes, it's definetely the `connectors`. I saw that you have pushed a fix, I'll test it later and let you know ;)

razerbann, 2020-04-15:
The problem seems to be fixed! Great!

commit 48165389 (2020-04-15): fix #5.