ibnuda/dactyl-keyboard #6: Dactyl Lightcycle: Some supports are missing

razerbann, 2020-04-15:
Using the default configuration on the web UI (except for the cutouts of the stabilizers that I enabled), I generated the following shape and it appears that some parts of the support connecting the wall to the plate are missing (pointed out by the arrows).

[image]

I'm pretty sure that the one in the middle is missing but I don't know about the one on the right and the one on the left.

ibnuda, 2020-04-15:
for the thumb-front-wall (left) and thumb-back-wall (right), it's fixed in the previous commit. for the thumb-left-wall (middle) i don't think it does matter that much.

razerbann, 2020-04-15:
> for the thumb-front-wall (left) and thumb-back-wall (right), it's fixed in the previous commit. for the thumb-left-wall (middle) i don't think it does matter that much.

I didn't check your last commit, I'll check if the problem is solved ;)
For the piece missing from the thumb-left-wall, while it's true that it doesn't matter that much, but it would be great though to add it ;)

closed 2020-04-18 by razerbann