ibnuda/dactyl-keyboard #13: Dactyl thumb cluster with 8 keys

CreamyCookie, 2020-06-06:
First of all: Wow, this is one of the most amazing things I've seen in the MK community. Thanks for creating this!

My request is about the thumb cluster. I personally think the thumb cluster is really useful in Dactyl-style keyboards, but my experience with the 2u keys is that they're great, but would basically provide twice the benefit by splitting them up. I usually only hit the top half of the 2u key, so having another key below would be really useful for me. It would also allow for a better thumb curvature to make all the thumb keys easier to reach.

[image]

(*Image source:* https://github.com/adereth/dactyl-keyboard/blob/master/README.md)

ibnuda, 2020-06-07:
OK. I will try to do that. should be doable.

ibnuda, 2020-06-07:
@CreamyCookie, please check the site again. it should be able to generate 8 thumb keys, now. and if you're ok with it, please close the issue.

thanks.

CreamyCookie, 2020-06-07:
@ibnuda Thanks so much! I haven't (yet) build a single keeb (using Kinesis Advantage atm) so I can't fully evaluate it. That said, the thumb cluster is curved and it looks good to my untrained eye :)

Really cool, thank you!

closed 2020-06-07 by CreamyCookie

CreamyCookie, 2020-06-07:
Pictures from OpenSCAD

[image]
[image]
[image]