ibnuda/dactyl-keyboard #15: Manuform with Lightcycle thumb cluster

CreamyCookie, 2020-06-11:
While I'm personally not a fan of Manuform's spaced out thumb key positions, I really like the thumb cluster angle. (It makes a lot of sense since, of course, our thumbs are opposable.)

[image]

Please forgive the terrible edit :smile: 

*(Image source: http://blog.pzs.me/posts/dactyl-manuform-log/index.html)*

ibnuda, 2020-06-14:
i will see what i can come up with.

CreamyCookie, 2020-06-15:
@ibnuda Thanks so much for these quick fixes / changes!

Could it be that the web generator is not yet updated? I can select a negative angle but the resulting file has none and the overlap from #18 is still there.

This is with -20 thumb angle:

 [image]

[Settings as always (just in case)](https://user-images.githubusercontent.com/3063858/84699490-e35dab80-af51-11ea-8f90-c27067a17396.png)

ibnuda, 2020-06-16:
ah, you're right. I didn't touch the handler function. I will fix it in a few hours.

CreamyCookie, 2020-06-16:
Okay, I'll try again later! =)

ibnuda, 2020-06-16:
these following images use your setting.

[image]

[image]

while the following images use pi/-9 for thumb tenting angle. i also have put it live.

[image]
[image]

CreamyCookie, 2020-06-16:
First of all, thank you. There are a couple issues (I'll number them for easy reference), but I like this. =)

1. The artifact is still there (but maybe it's not a problem?): 

[image]

---

2. Do you think the two keycaps between thumb cluster and main area could intersect? (There's no keycaps option in the Lightcycle settings to test this.)  

[image]

 At first, it seems that moving the thumb cluster away would solve that, but I think that would be problematic, since the thumb cluster is already a bit too far away.

Crazy idea but would probably be too much effort: 

[image]

---

3. Could this small protrusion in green be removed?

[image]

----

I think an increased thumb curvature would make this really really good. To clarify #14 I've added a small photoshop edit.

ibnuda, 2020-06-17:
thumb column curvature: pi/9
thumb row curvature: pi/9
thumb-tenting-angle: pi/9
[image]

thumb column curvature: pi/9
thumb row curvature: pi/9
thumb-tenting-angle: pi/-9
[image]

and the artifact in pic 1 & 3 are not there anymore.

CreamyCookie, 2020-06-17:
Fantastic job. Thanks a ton! =D

My current favorite ([settings](https://user-images.githubusercontent.com/3063858/84950515-59e3e000-b0ef-11ea-90ef-75410dbdfcd2.png)): 
[image]

closed 2020-06-17 by CreamyCookie