ibnuda/dactyl-keyboard #19: Ring finger key intersects with border

CreamyCookie, 2020-06-17:
The area where the keycap would be seems to intersect with the border / rim:

[image]

Zoomed out: 
[image]

[settings](https://user-images.githubusercontent.com/3063858/84950515-59e3e000-b0ef-11ea-90ef-75410dbdfcd2.png)

If it's not too hard to do, I'd love an option to disable the whole border to create a "floating key design", though I guess that would almost turn it into a manuform. Might look cool.

ibnuda, 2020-06-18:
no-border:
[image]

with-border:
[image]

are you sure you want to use that kind of setting? if so, please check the site.

CreamyCookie, 2020-06-18:
"to border or not to border" love it  :smile: 

I've created a new issue for the borderless idea and added some feedback: #20

Thanks for the the option and the fix!

closed 2020-06-18 by CreamyCookie