ibnuda/dactyl-keyboard #21: JSON Load Button

CreamyCookie, 2020-06-19:
Just saw that there's now a JSON export button. You probably planned this already, but I think it would be cool if there was a <kbd>Load JSON</kbd> button at the top that did a `prompt("Please enter your JSON Config")` which would then get passed to a function that set all the form elements to the contained values.

That would make it really easy to share configs (especially in bug reports).

I (re)learned a bit of Javascript recently and came up with this, which works with a flat JSON:

```javascript
jsonStringToForm('{"thumb-count": "8"}', document.getElementsByTagName("form")[0])
```

but not with the JSON that's exported, so it still needs some work:

```javascript
// LICENSE: public domain

const SIMPLE_FORM_TYPES = new Set(["select-one", "text", "textarea", "number", "range"]);

function isSimpleValueFormElement(el) {
    return SIMPLE_FORM_TYPES.has(el.type);
}

function jsonStringToForm(jsonString, formElement) {
    const obj = JSON.parse(jsonString);
    
    [...formElement.elements].forEach((el) => {
        if (!el.name || !el.value) return;
        
        var val = obj[el.name];
        
        if (el.type === "checkbox") {
            if (Array.isArray(val) && val.includes(el.value)) {
                el.checked = true;
            }
        } else if (isSimpleValueFormElement(el)) {
            el.value = val;
        }
    });
}
```

ibnuda, 2020-06-22:
it is done, i think. please check the site.

CreamyCookie, 2020-06-22:
It does seem to have worked for many (all of the ones in `keys` worked) but not all attributes:

Lightcycle input with every field differing from the default:

```json
{"keys":{"columns":7,"num-row":false,"last-row":false,"thumb-count":"eight"},"curve":{"alpha":20,"beta":30,"tenting":8,"thumb-alpha":20,"thumb-beta":32,"thumb-tenting":14},"connector":{"external":false},"form":{"hotswap":true,"thumb-offset-x":56,"thumb-offset-y":57,"thumb-offset-z":39,"z-offset":24,"manuform-offset":true,"border":false,"thick-wall":true},"misc":{"right-side":false,"screw-inserts":true}}
```

and this is what was set and exported:

```json
{"keys":{"columns":7,"num-row":false,"last-row":false,"thumb-count":"eight"},"curve":{"alpha":20,"beta":30,"tenting":12,"thumb-alpha":36,"thumb-beta":36,"thumb-tenting":24},"connector":{"external":false},"form":{"hotswap":false,"thumb-offset-x":52,"thumb-offset-y":45,"thumb-offset-z":27,"z-offset":18,"manuform-offset":false,"border":true,"thick-wall":false},"misc":{"right-side":true,"screw-inserts":false}}
```

Diff (left is input):

[image]

ibnuda, 2020-06-23:
you're right. i forgot to change some of the inputs' ids. please check the site.

CreamyCookie, 2020-06-23:
Nice! Nearly all fields work. I think the only ones not working for Lightcycle are:

* Use loligagger's external holder or default cutout (RJ9 + big USB hole)?
* Show Keycaps?

and for Manuform: Connector, USB Hole, Stagger

I'm getting an exception on export (only for Lightcycle):

```
java.lang.NumberFormatException: null
	at java.base/java.lang.Integer.parseInt(Integer.java:614)
	at java.base/java.lang.Integer.parseInt(Integer.java:770)
	at dactyl_keyboard.handler$parse_int.invokeStatic(handler.clj:15)
	at dactyl_keyboard.handler$parse_int.invoke(handler.clj:14)
	at dactyl_keyboard.handler$generate_lightcycle.invokeStatic(handler.clj:148)
	at dactyl_keyboard.handler$generate_lightcycle.invoke(handler.clj:135)
	at compojure.response$eval1946$fn__1947.invoke(response.clj:47)
	at compojure.response$eval1868$fn__1869$G__1859__1876.invoke(response.clj:7)
	at compojure.core$wrap_response$fn__3823.invoke(core.clj:158)
	at compojure.core$wrap_route_middleware$fn__3807.invoke(core.clj:128)
	at compojure.core$wrap_route_info$fn__3812.invoke(core.clj:137)
	at compojure.core$wrap_route_matches$fn__3816.invoke(core.clj:146)
	at compojure.core$routing$fn__3831.invoke(core.clj:185)
	at clojure.core$some.invokeStatic(core.clj:2701)
	at clojure.core$some.invoke(core.clj:2692)
	at compojure.core$routing.invokeStatic(core.clj:185)
	at compojure.core$routing.doInvoke(core.clj:182)
	at clojure.lang.RestFn.applyTo(RestFn.java:139)
	at clojure.core$apply.invokeStatic(core.clj:667)
	at clojure.core$apply.invoke(core.clj:660)
	at compojure.core$routes$fn__3835.invoke(core.clj:192)
	at ring.middleware.json$wrap_json_body$fn__7005.invoke(json.clj:58)
	at ring.middleware.json$wrap_json_response$fn__7058.invoke(json.clj:139)
	at ring.middleware.keyword_params$wrap_keyword_params$fn__5597.invoke(keyword_params.clj:36)
	at ring.middleware.params$wrap_params$fn__4063.invoke(params.clj:67)
	at ring.middleware.absolute_redirects$wrap_absolute_redirects$fn__6041.invoke(absolute_redirects.clj:47)
	at ring.middleware.content_type$wrap_content_type$fn__5989.invoke(content_type.clj:34)
	at ring.middleware.default_charset$wrap_default_charset$fn__6013.invoke(default_charset.clj:31)
	at ring.middleware.not_modified$wrap_not_modified$fn__5970.invoke(not_modified.clj:53)
	at clojure.lang.Var.invoke(Var.java:384)
	at ring.adapter.jetty$proxy_handler$fn__472.invoke(jetty.clj:27)
	at ring.adapter.jetty.proxy$org.eclipse.jetty.server.handler.AbstractHandler$ff19274a.handle(Unknown Source)
	at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:127)
	at org.eclipse.jetty.server.Server.handle(Server.java:500)
	at org.eclipse.jetty.server.HttpChannel.lambda$handle$1(HttpChannel.java:386)
	at org.eclipse.jetty.server.HttpChannel.dispatch(HttpChannel.java:562)
	at org.eclipse.jetty.server.HttpChannel.handle(HttpChannel.java:378)
	at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:270)
	at org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:311)
	at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:103)
	at org.eclipse.jetty.io.ChannelEndPoint$2.run(ChannelEndPoint.java:117)
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.runTask(EatWhatYouKill.java:336)
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.doProduce(EatWhatYouKill.java:313)
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.tryProduce(EatWhatYouKill.java:171)
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.run(EatWhatYouKill.java:129)
	at org.eclipse.jetty.util.thread.ReservedThreadExecutor$ReservedThread.run(ReservedThreadExecutor.java:388)
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:806)
	at org.eclipse.jetty.util.thread.QueuedThreadPool$Runner.run(QueuedThreadPool.java:938)
	at java.base/java.lang.Thread.run(Thread.java:834)
```

ibnuda, 2020-06-27:
it should've been fixed in [this commit](https://github.com/ibnuda/dactyl-keyboard/commit/cbf5c8ef32e736e91d71b338c3c88bd03307f5bd).

CreamyCookie, 2020-06-27:
@ibnuda I'll try again =)

CreamyCookie, 2020-06-27:
Lightcycle fully works (imported config with all fields different and it exported the exact same values)

Manuform's export is still broken. I've opened a new report (since this is for import): #30 

Thanks for the fixes!

closed 2020-06-27 by CreamyCookie