ibnuda/dactyl-keyboard #25: Rendered STL is non-manifold

Pimentoso, 2020-06-21:
Here is the json

```
{
    "keys": {
        "columns": 6,
        "rows": 5,
        "thumb-count": "three",
        "last-row": "full",
        "nubs": false,
        "alps": false,
        "inner-column": false
    },
    "curve": {
        "alpha": 12,
        "beta": 36,
        "centercol": 4,
        "tenting": 15
    },
    "connector": {
        "external": false,
        "trrs": true,
        "micro-usb": true
    },
    "form": {
        "hotswap": false,
        "stagger": true,
        "wide-pinky": false,
        "height-offset": 4,
        "wire-post": false,
        "screw-inserts": true
    },
    "misc": {
        "keycaps": false,
        "wrist-rest": false,
        "integrated-wrist-rest": false,
        "right-side": true,
        "case": true
    }
}
```

I open the generated scad file with openSCAD 2019.5, render it and export the stl. Meshmixer shows manifold issues.

PrusaSlicer 2.2.0 has problems slicing it

[image]

While it's all good in Cura 4.6.1 

[image]

It's not really an issue since I'm gonna use Cura anyways but just wanted to let you know.

dereknheiley, 2020-07-24:
Confirmed, also seeing this in prusa slicer.
Possible work arounds for the time being:
1. ideamaker (like cura) was able to "repair" the STL file that OpenScad generates. 
2. stl file can also be manually fixed with a site like https://service.netfabb.com/service.php

dereknheiley, 2020-07-29:
i was updating the thumb cluster locally and noticed the "needs repair" went away in idea maker after the latest export from scad to stl, i'l try and track down (one of?) the issues if i can

wez, 2021-02-24:
FWIW, I'm also seeing a combination of 175 non-manifold edges and 754 intersecting faces for my different config when uploading the stl to 3dhubs.com for printing.  I tried running it through netfabb but it does't fix everything.  I also tried Meshmixer but that hilariously auto-fixed the model into something resembling a plate of mashed potatoes!  I also tried the Prusa slicer but that didn't really fix anything. Is there something that could be done in generation to avoid creating the problematic geometry?