ibnuda/dactyl-keyboard #26: Lightcycle: Option for wall thickness

CreamyCookie, 2020-06-21:
If not too time consuming, I think it would be cool to have a thickness option for the wall, to give the lightcycle an even sturdier feel.

**Comparision at the same zoom level**
[image]

[image]

ibnuda, 2020-06-22:
i made an option to use `thick-wall` which 3 mm thick. please check the site.

CreamyCookie, 2020-06-22:
@ibnuda Thank you! I think 3 mm might be a bit too much. Maybe 2? (light green)

[image]

CreamyCookie, 2020-06-22:
@ibnuda Thank you! I think 3 mm might be a bit too much. I've marked 2mm in light green:

[image]

closed 2020-06-23 by CreamyCookie