ibnuda/dactyl-keyboard #28: Manuform: Thumb curvature

CreamyCookie, 2020-06-21:
I'm starting to like the Manuform design more and more, so I'd love to have the thumb curvature there as well (blue line to indicate a curvature that seems more optimal to me):

[image]

georgeinmexico, 2020-08-14:
Agreed. I would like to be able to drop the thumb cluster closer to the desk.

happysalada, 2021-04-10:
I have a PR for this. If you have time to try it, I would be interested in your feedback.
https://github.com/ibnuda/dactyl-keyboard/pull/80
I've taken the approach that you can define an offset and a tenting for each of the thumb key.
I've only done it for the top 3 thumb keys, but you can essentially put it into any shake you want.
The angles are in degrees, which I think is more intuitive.
Let me know.

CreamyCookie, 2021-04-10:
@happysalada Very cool! Thank you for doing this!

> If you have time to try it, I would be interested in your feedback.

I have looked a bit through the pull request and code but I don't have time for testing it at the moment. Would love to give feedback though! Testing would easier via a web interface though.. (I haven't used Clojure yet and even then it seems easier to configure via an UI.)

>  The angles are in degrees, which I think is more intuitive.

Definitely agree!

happysalada, 2021-05-30:
The PR for custom thumb placement has just been merged.
Unfortunately the UX is not just defining a curvature.
You can define for thumbs, a position and an rotation (the rotation is in degrees, be careful).
This enables you to do anything actually, not just curvature.
Not all the thumbs are customisable, only the top 3.
Adding the rest is relatively straightforward (I only use 3 thumb keys, so didn't go all the way).

yejianfengblue, 2021-06-27:
@happysalada Could you please share your manuform params? With all the default value of [manuform](https://dactyl.siskam.link/manuform) and `Thumb Key Count = 3` and `Custom thumb cluster (Experimental) = Yes`, the thumb cluster is weird.

```json
{
  "keys": {
    "columns": 5,
    "rows": 4,
    "thumb-count": "three",
    "last-row": "two",
    "switch-type": "box",
    "inner-column": "normie",
    "hide-last-pinky": false
  },
  "curve": {
    "alpha": 12,
    "pinky-alpha": 12,
    "beta": 36,
    "centercol": 4,
    "tenting": 15,
    "rotate-x": 180
  },
  "connector": {
    "external": false,
    "type": "rj9",
    "micro-usb": false
  },
  "form": {
    "stagger-pinky-y": -13.0,
    "thumb-cluster-offset-x": 6.0,
    "hotswap": false,
    "stagger-index-y": 0.0,
    "thumb-middle-left-tenting-y": 69.0,
    "thumb-top-left-offset-y": -16.0,
    "wire-post": false,
    "stagger": true,
    "stagger-ring-z": 0.0,
    "thumb-top-right-tenting-y": 69.0,
    "stagger-middle-z": -6.5,
    "screw-inserts": false,
    "thumb-top-left-tenting-z": 69.0,
    "thumb-top-left-tenting-x": 69.0,
    "thumb-middle-left-offset-x": -53.0,
    "stagger-middle-y": 2.8,
    "thumb-middle-left-offset-z": -12.0,
    "thumb-middle-left-offset-y": -26.0,
    "height-offset": 4,
    "thumb-middle-left-tenting-x": 69.0,
    "wide-pinky": false,
    "thumb-top-right-offset-x": -15.0,
    "stagger-index-z": 0.0,
    "thumb-top-right-tenting-z": 69.0,
    "thumb-cluster-offset-z": 7.0,
    "custom-thumb-cluster": true,
    "thumb-top-right-tenting-x": 69.0,
    "stagger-ring-y": 0.0,
    "thumb-middle-left-tenting-z": 69.0,
    "thumb-top-left-tenting-y": 69.0,
    "thumb-top-right-offset-y": -10.0,
    "thumb-top-right-offset-z": -5.0,
    "thumb-top-left-offset-x": -35.0,
    "thumb-cluster-offset-y": -3.0,
    "thumb-top-left-offset-z": -2.0,
    "stagger-pinky-z": 6.0
  },
  "misc": {
    "keycaps": false,
    "right-side": true
  }
}
```
[image]

happysalada, 2021-06-28:
```
{"keys":{"columns":5,"rows":4,"thumb-count":"three","last-row":"zero","switch-type":"choc","inner-column":null,"hide-last-pinky":false},"curve":{"alpha":8,"pinky-alpha":8,"beta":69,"centercol":2,"tenting":4,"rotate-x":-6},"connector":{"external":false,"type":"none","micro-usb":true},"form":{"stagger-pinky-y":-20.0,"thumb-cluster-offset-x":12.0,"hotswap":true,"stagger-index-y":-10.0,"thumb-middle-left-tenting-y":-45.0,"thumb-top-left-offset-y":-10.0,"wire-post":false,"stagger":true,"stagger-ring-z":10.0,"thumb-top-right-tenting-y":-45.0,"stagger-middle-z":0.0,"screw-inserts":false,"thumb-top-left-tenting-z":0.0,"thumb-top-left-tenting-x":25.0,"thumb-middle-left-offset-x":-47.0,"stagger-middle-y":0.0,"thumb-middle-left-offset-z":-20.0,"thumb-middle-left-offset-y":-10.0,"height-offset":16,"thumb-middle-left-tenting-x":25.0,"wide-pinky":false,"thumb-top-right-offset-x":-12.0,"stagger-index-z":10.0,"thumb-top-right-tenting-z":-15.0,"thumb-cluster-offset-z":5.0,"custom-thumb-cluster":true,"thumb-top-right-tenting-x":25.0,"stagger-ring-y":-10.0,"thumb-middle-left-tenting-z":15.0,"thumb-top-left-tenting-y":-45.0,"thumb-top-right-offset-y":-10.0,"thumb-top-right-offset-z":15.0,"thumb-top-left-offset-x":-30.0,"thumb-cluster-offset-y":0.0,"thumb-top-left-offset-z":-2.0,"stagger-pinky-z":30.0},"misc":{"keycaps":false,"right-side":true}}
```

Those are my params, I can post an image if need be.
Maybe the default params need to be updated?

yejianfengblue, 2021-06-28:
Thanks for the params. It's a good reference. I think the default value of xyz deg of top right, top left, middle left should be set to a more suitable value rather than `Default (without curvature)`. It's quit hard to find a suitable xyz deg combination without any clue.

happysalada, 2021-06-28:
If you have a suggestion for the default params, I'm happy to make a PR. I just really had no idea what would be good defaults.
I only use three keys on the thumbs and so I was afraid my values would completely break the other cases of 4 or 5 thumb keys.