ibnuda/dactyl-keyboard #30: Manuform export doesn't work for all fields

CreamyCookie, 2020-06-27:
After changing every field, so the config looks like this:

[image]

This was exported

`{"keys":{"columns":4,"rows":3,"thumb-count":"four","last-row":"zero","switch-type":"alps","inner-column":true},"curve":{"alpha":14,"beta":30,"centercol":1,"tenting":11},"connector":{"external":false,"trrs":false,"micro-usb":false},"form":{"hotswap":true,"stagger":true,"wide-pinky":true,"height-offset":10,"wire-post":true,"screw-inserts":true},"misc":{"keycaps":true,"wrist-rest":false,"integrated-wrist-rest":true,"right-side":false,"case":true}}`

None of the "connector" match and the "stagger" should be false.

ibnuda, 2020-06-28:
it should've been fixed and live at the site. sorry for these bugs. i always open the project past midnight.

CreamyCookie, 2020-06-28:
Thanks! No problem. Finding time for these projects can be hard sometimes.

closed 2020-06-28 by CreamyCookie