ibnuda/dactyl-keyboard #32: Manuform: Gapless 7-key thumb cluster

CreamyCookie, 2020-06-28:
Having looked a bit into the source code (without Clojure knowledge), the thumb cluster of the Manuform is more hard-coded than the Lightcycle, which seems to make #28 very time consuming to implement. 

Additionally, curvature for the thumb cluster doesn't fully address a problem that I feared and have now seen in a lot of reviews (examples: [1](https://www.reddit.com/r/ErgoMechKeyboards/comments/d89rkz/help_want_to_build_dactylmanuform_but_which_is/f196zc7/),  [2](https://www.reddit.com/r/MechanicalKeyboards/comments/ajgdg5/criticisms_of_dactylmanuform_dacman_from_another/), [3](https://www.reddit.com/r/MechanicalKeyboards/comments/9d0gqn/just_finished_my_first_dactyl_manuform_i_cant/e5ey4d4/), [4](https://www.reddit.com/r/ErgoMechKeyboards/comments/hfug5k/how_to_best_hit_the_middle_thumb_key_of_a_dactyl/fw1bi1g/)) :  the outer thumb keys being too far away (which is partially a result of the horizontal rotation, which requires gaps and distance)

So, it seems a layout where keys are next to each other, like the following 7-key layout would be really useful to have:

[image]

The ones marked with `(tented)` would be tented toward the center of the two leftmost keys, to make them easier to reach. It would kind of like this when looking down from above the key **1**:

[image]

(the outer row would be tented the same amount from the middle row, as the middle row is tented towards the leftmost row)

The reason for the shift in the last row is that this keeps much of the optimal curve (marked in green here):

[image]

And here's the layout added on top of the Manuform layout (just for fun):

[image]

dereknheiley, 2020-07-09:
my next version will have the mini 5-thumb cluster, since i also find the stock dactyl-manuform thumb buttons to not be in the best position, that said, the dactyl, and similar kinesis and ergo-dox thumb clusters are pretty busy with that many keys in close proximity, have you tried simulating this with a key switch tester or some creative hot glue-ing?

CreamyCookie, 2020-07-09:
Haven't tested it yet but I use the Kinesis Advantage atm. which has a 6-key thumb cluster with relative close proximity between keys and I actually find the outer keys (alt, home, end) too far away. <kbd>alt</kbd> is a tiny bit worse than the other two) since you have to move the whole hand so your thumb can reach it. 
(Of course this depends on your hand, but mine are  probably a bit larger than average.)

[image]

Another thing I've noticed with the Advantage is that I almost only hit the two 2u keys at the top, so given the right angle, having another two keys would be nice. That said, the lower half of the Advantage spacebar key might not be that comfortable to hit, so a 6-key thumb cluster with tenting would just be fine. 

**Some ideas:**

[image]

Also from having read and watched a lot of reviews of both the Advantage and the Dactyl (Manuform and not), I'd say that the thumb cluster seems to be the area of those keyboards that has the biggest potential for improvement. A lot of reviews mention they're not quite happy with the thumb cluster. 

The Dactyl also has a downwards slope for the thumb cluster, and some reviews (I agree with) mentioned that it doesn't seem to make sense that the thumb keys are curving away from your hand, rather than curving towards you as is the case for the main letter keys.

One keyboard that does that is the [Tightyl](https://github.com/okke-formsma/dactyl-manuform-tight) by @okke-formsma (also based on Dactyl by @adereth):

[image]

*(image source: https://imgur.com/a/YePWDY5#HrvDgbh)*

dereknheiley, 2020-07-11:
yeah, it would be nice to try a / have the ability to change to a concave version of, the "5 (Mini)"