ibnuda/dactyl-keyboard #33: Configurable middle finger height offset

CreamyCookie, 2020-06-29:
Something that I've only noticed just now is that the height offset of the middle finger is actually a bit too extreme (at least for me), so it would be cool if it was configurable. 

[image]

[image]

When I let my fingers hover over the keyboard the difference between ring and middle finger is actually pretty minor. Not exactly the  position I'm talking about but this image shows what I mean:

[image]

While not perfect, here's the Advantage again for reference:

[image]
[image]

ibnuda, 2020-07-08:
should be fixed already. please check the site.

CreamyCookie, 2020-07-08:
@ibnuda Awesome. Thank you once again! =D

closed 2020-07-08 by CreamyCookie