ibnuda/dactyl-keyboard #36: add preview render

dereknheiley, 2020-07-05:
i think someone else mentioned this, but there doesn't appear to be an issue / feature request for integrating a https://openjscad.org/ view somewhere on the page

Add support for showing a preview of the design which stays on the page to support the user with the following series of tasks:
1. select a default
2. preview default design
3. update some parameters such as number of columns, or removing pinky bottom row, etc.
4. see updated render
5. tweak some more things to see what they do
6. see updated render
7. tweak final settings to get it just right
8. finally happy with updated render
9. download .scad file to export to stl for printing

CreamyCookie, 2020-07-08:
That would be awesome, but I think it would probably tax the server too much..

Maybe it would be possible to compile the Clojure code to JavaScript using something like https://github.com/clojure/clojurescript :smile:

rianadon, 2023-03-03:
I'm using OpenJSCAD in my configurator: https://github.com/rianadon/dactyl-configurator/, which might serve as inspiration. OpenJSCAD does a fairly great job generating previews, but it sadly doesn't do as great of a job at generating previews as OpenSCAD (the renders tend to have more missing faces).

CreamyCookie, 2023-06-10:
@rianadon Very cool what you did! Look forward to see what Cosmos will look like :+1: