ibnuda/dactyl-keyboard #40: Sometimes connector tray not fully connected

CreamyCookie, 2020-07-08:
[image]

Could be unrelated, but I think this happens mostly when "Center of the curvature for the column" is pinky or ring finger.

`{"keys":{"columns":6,"rows":5,"thumb-count":"five","last-row":"two","switch-type":"box","inner-column":false,"hide-last-pinky":false},"curve":{"alpha":12,"pinky-alpha":8,"beta":18,"centercol":3,"tenting":6},"connector":{"external":false,"trrs":true,"micro-usb":false},"form":{"stagger-pinky-y":-12.0,"hotswap":true,"stagger-index-y":0.0,"wire-post":true,"stagger":true,"stagger-ring-z":0.0,"stagger-middle-z":-4.5,"screw-inserts":true,"stagger-middle-y":2.8,"height-offset":16,"wide-pinky":false,"stagger-index-z":0.0,"stagger-ring-y":0.0,"stagger-pinky-z":4.0},"misc":{"keycaps":true,"right-side":true,"case":true}}`

ibnuda, 2020-07-11:
shouldn't be a problem anymore. moved the connectors to the center a bit more. please check site.

CreamyCookie, 2020-07-11:
Looks great. Thanks!

closed 2020-07-11 by CreamyCookie