ibnuda/dactyl-keyboard #41: Add part description suffix to filename

dereknheiley, 2020-07-09:
It would also be very helpful to have the filename suffixed with the part description, like "Left", "Right", "Plate"

For example
```
manuform-left-2020.07.09-02.37.23.scad
manuform-right-2020.07.09-02.37.19.scad
manuform-plate-2020.07.09-02.36.45.scad
```

ibnuda, 2020-07-11:
should be fixed already. please check the site.

dereknheiley, 2020-07-11:
the case-left and case-right work, but i guess the plate should also have the -right and -left suffix too, i didn't really consider that

so,
manuform-case-left-2020.07.09-02.37.23.scad
manuform-case-right-2020.07.09-02.37.19.scad
manuform-plate-left-2020.07.11-04.36.18.scad
manuform-plate-right-2020.07.11-04.36.14.scad

ibnuda, 2020-07-11:
the plate is reversible. just print it twice and then flip one.

closed 2020-07-15 by dereknheiley