ibnuda/dactyl-keyboard #43: Manuform: Import doesn't set all fields

CreamyCookie, 2020-07-11:
Haven't tested this for Lightcycle, but there seem to be more problems with the import. Just noticed that my current config was missing a lot of settings on import.

What I imported:

```json
{"keys":{"columns":6,"rows":5,"thumb-count":"five","last-row":"two","switch-type":"box","inner-column":false,"hide-last-pinky":false},"curve":{"alpha":12,"pinky-alpha":8,"beta":18,"centercol":3,"tenting":6},"connector":{"external":false,"trrs":true,"micro-usb":false},"form":{"stagger-pinky-y":-12.0,"hotswap":true,"stagger-index-y":0.0,"wire-post":true,"stagger":true,"stagger-ring-z":0.0,"stagger-middle-z":-4.5,"screw-inserts":true,"stagger-middle-y":2.8,"height-offset":16,"wide-pinky":false,"stagger-index-z":0.0,"stagger-ring-y":0.0,"stagger-pinky-z":4.0},"misc":{"keycaps":true,"right-side":true,"case":true}}
```

What was set and later exported:

```json
{"keys":{"columns":6,"rows":5,"thumb-count":"five","last-row":"two","switch-type":"box","inner-column":false,"hide-last-pinky":false},"curve":{"alpha":12,"pinky-alpha":8,"beta":18,"centercol":3,"tenting":6},"connector":{"external":false,"trrs":true,"micro-usb":false},"form":{"stagger-pinky-y":-13.0,"hotswap":false,"stagger-index-y":0.0,"wire-post":false,"stagger":true,"stagger-ring-z":0.0,"stagger-middle-z":-6.5,"screw-inserts":false,"stagger-middle-y":2.8,"height-offset":4,"wide-pinky":false,"stagger-index-z":0.0,"stagger-ring-y":0.0,"stagger-pinky-z":6.0},"misc":{"keycaps":false,"right-side":true,"case":true}}
```

Diff:

[image]

happysalada, 2020-10-18:
I believe this was fixed in https://github.com/ibnuda/dactyl-keyboard/pull/51
Feel free to post an update if something is missing.

CreamyCookie, 2020-10-18:
Just tested. Yep, seems to be fixed!

closed 2020-10-18 by CreamyCookie