ibnuda/dactyl-keyboard #45: Manuform: 6th row key caps get stuck against case when using column stagger

dereknheiley, 2020-07-15:
Keys, 8 and 9 (and 2 and 3) just barely have clearance, and keys F8 and F9 (and F2 and F3) rub and get stuck against the case on a 6th row when column stagger is used.

Suggest enforcing minimum clearance around keys, and forcing outer three columns to fan out slightly at the top to ensure minimum wall thickness is maintained.

[image]

CreamyCookie, 2020-07-18:
I could be mistaken, but I think that's related to #39 (which I opened). The vertical center of the curvature seems to be at the top (rather than somewhere in the middle) and thus causes the keys to collide at that point. 

Of course, one could just increase the distance at the top, but I'm not sure if that would be preferable, given that the goal for ergonomics usually is to reduce unnecessary finger movement.

dereknheiley, 2020-07-29:
ah, i'm realizing now (poking through the code trying to fix it myself) that the problem is because the underlying geometry i want is impossible. IE: you can't have a very deep row curve, and not run into collisions like this on the first and last rows of the columns when using aggressive column stagger / recess

closed 2020-07-29 by dereknheiley

CreamyCookie, 2020-07-29:
Yes, that's true, but again, it seems to me that at least in many cases, this is caused by the "Center of the curvature for the row" being at the top, not the y-center, no?

Again, maybe I'm mistaken, but at least I don't see why else here on the right the collision is at the top.

[image]

Of course, it's possible that if the center was different it would instead collide at both the top and bottom, but