ibnuda/dactyl-keyboard #47: Not enough space for keycaps in default 6 row configuration.

georgeinmexico, 2020-08-14:
Thank you for such an excellent tool!

Using the web generator, I leave everything default except for "show keycaps" and Rows: 6. The resulting Scad file doesn't have enough space to accommodate standard keycaps and keypresses on the top row (function row) of the ring finger and middle finger. 
Is there a parameter I can change to make more space for those top keys?

Thanks again for making this great tool free.

georgeinmexico, 2020-08-14:
[image]
I forgot to attach the picture. I can confirm that the keys don't have space to travel on my printed version of the board.

dereknheiley, 2020-08-17:
i've been experimenting with different row curvatures and middle, ring, and pinky finger staggers to mitigate the issue so far
if you have the code downloaded and running via `lein repl` you could use a config option such as `:configuration-beta                   (/ pi 50000)` to flatted out the row curvature and give the keys more room

[image]

georgeinmexico, 2020-08-19:
Thank you for your help with this. Thank you for giving me a CLI solution. 

I am using the web configuration tool http://dactyl.siskam.link/manuform . If the issue gets mitigated in the online tool, please let me know!

dereknheiley, 2020-08-29:
what config options are you using ? i can generate the STL file for you if you want with an almost flat row curve as i have pictured above

georgeinmexico, 2020-09-04:
You are very generous to offer to do that! Below is the json export of my current config.

I have the maximum pinky column curvature set. Could you take a look at the STL file after you generate it? There may not be enough space in the case to wire the top row of switches. Thank you!

{"keys":{"columns":8,"rows":6,"thumb-count":"five","last-row":"two","switch-type":"mx","inner-column":true,"hide-last-pinky":false},"curve":{"alpha":9,"pinky-alpha":5,"beta":36,"centercol":4,"tenting":15},"connector":{"external":true,"trrs":true,"micro-usb":true},"form":{"stagger-pinky-y":-13.0,"hotswap":true,"stagger-index-y":0.0,"wire-post":false,"stagger":true,"stagger-ring-z":0.0,"stagger-middle-z":-6.5,"screw-inserts":true,"stagger-middle-y":2.8,"height-offset":4,"wide-pinky":true,"stagger-index-z":0.0,"stagger-ring-y":0.0,"stagger-pinky-z":6.0},"misc":{"keycaps":true,"right-side":true,"case":true}}

dereknheiley, 2020-09-05:
[image]

dereknheiley, 2020-09-05:
i had to adjust the pinky curvature a bit to make the case not run through the socket area (7 instead of 5), and increase the height (14 instead of 4) so that the outer pinky row was completely rendering / printing with the flatter curvature (50,000 instead of 36).

Also, are you sure you wanted 8 columns? Seems like 2 to many.

I'd recommend printing the test STL file to verify that a couple of critical areas are usable before printing the entire model. The keycaps might still be a little tight depending on the exact type and tolerances of them.
[8col_with_STLs.zip](https://github.com/ibnuda/dactyl-keyboard/files/5178200/8col_with_STLs.zip)

georgeinmexico, 2020-09-08:
These files are super helpful, Thank you. How did you generate the "Test.stl" file? I have been using openSCAD to view the scad files, but I don't know how to crop the file to print only a portion of it.

dereknheiley, 2020-09-09:
Ideamaker (the 3d slicer software) allows you to cut STL files into pieces multiple times. I think Prusa Slicer / Slic3r, and there 3d printing slicer software has similar functionality.

dereknheiley, 2020-09-09:
note: i just realized that i slightly modified one of the thumb buttons closest to the alpha keys, and also, as you can see the hot swap holders are a slightly different version i haven't merged into the repo yet

ibnuda, 2020-11-07:
in `common.clj` change `extra-width` from  `2.5` to `3.5`.