ibnuda/dactyl-keyboard #48: Feature request: adjust altitude of the 5 mini thumb cluster

georgeinmexico, 2020-09-04:
The five key "mini" thumb cluster has enough space underneath the keys that it could be lowered by up to 1cm without the wires underneath bumping the bottom. Could you add parameter to allow the thumb cluster to sit lower down? (closer to the baseplate).

dereknheiley, 2020-09-05:
that's an interesting idea, although keep in mind that the distance to the lower thumb buttons will increase in a less desirable way compared to the inner thumb buttons that might seem to be in a better position when they're lower

happysalada, 2020-10-18:
@georgeinmexico if you're not scared of running this yourself, you can adjust the parameter here
https://github.com/ibnuda/dactyl-keyboard/blob/refaktor/src/dactyl_keyboard/manuform.clj#L15
I'm currently experimenting on this and trying to find out what makes the most sense.
I'll update if I find out new things.

georgeinmexico, 2020-10-19:
Thank you for adding this feature. I am scared of running the build environment myself, as it relies on several technologies that I don't have an environment set up for. 

I saw that the thumb adjustments are appearing in the Web GUI for light cycle (but not dactyl). They might be causing the light cycle to give an error 500 in the web app.

Blessings,
George

happysalada, 2020-10-19:
They are not on the webgui yet.
Let me make a coutple of prints to see how it affects things before putting it on the webUI.
I'll update here when it's ready.
I'm only working on the dactyl though, I hope that's the one you are looking for.

georgeinmexico, 2020-10-20:
Yes, please, I'm looking forward to seeing the thumb cluster altitude adjustment on the dactyl!

Even though I don't have your technical skills, I would love to contribute to your project however I can. May I suggest adding the words "From" and "Than" to the labels for the thumb cluster height:

Thumb Position in X direction (left/right)
Further from Alpha Keys
Closer to Alpha Keys
Even Closer Closer Closer to Alpha Keys

Thumb Position in Y direction (front/back)
Farther from keyboard body
Even Farther from body

Thumb Position in Z direction (up/down)
Lower than Body
Even Lower than Body
Higher than keyboard body.

happysalada, 2020-10-31:
I've got a PR in, a little more patience and it should be merged.

Of course, when you test it, if you have any suggestions/comments, feel free to share, I can make some adjustments if needed.

closed 2020-10-31 by ibnuda