ibnuda/dactyl-keyboard #59: Reduce thickness of case

RaoulDuke-Esq, 2020-10-15:
I would like to reduce the overall thickness (depth) of the shell of the Manuform variant because I find the case to be too thick to accommodate Kailh switches or to use Amoeba PCBs because the case is so thick that the pins on the switches can't reach the PCBs.  I found "plate-thickness" in common.clj which can be adjusted so Kailh switches actually clip in properly, but this does not solve the problem of the overall socket being too deep.

I believe that reducing the thickness would also help with #46 - the external holders do not fit because they are not made to accommodate the extra thick walls of this case.

Where can I adjust the thickness of the walls/case?

happysalada, 2020-10-18:
If I understand you correctly, I have a similar problem.
Currently I am experimenting with the "web-thickness" parameter
https://github.com/ibnuda/dactyl-keyboard/blob/refaktor/src/dactyl_keyboard/common.clj#L367
That will reduce the thickness of the plate.
However, it won't help you with the socket thickness.

To reduce the thickness of the socket you have to play with the params over here
https://github.com/ibnuda/dactyl-keyboard/blob/refaktor/src/dactyl_keyboard/common.clj#L266
When I added choc switches, that's what I changed. You can adjust the bottom and the top of the socket wall to adjust the thickness.
I'm not sure if it makes perfect sense. If you try a couple of values, it will make sense I believe.

happysalada, 2020-10-31:
I just issued a PR to fix this problem for the choc-switch.
https://github.com/ibnuda/dactyl-keyboard/pull/65
Once the PR is merged, can you check and let me know if that is the problem you're having with the kailh switches?

aabbett, 2020-10-31:
Will do!

c10l, 2020-11-10:
Has anyone been able to test it with Kailh? If so, which switch profile is the one that works, choc?

Alternatively, would someone be able to implement the solution in this PR? https://github.com/tshort/dactyl-keyboard/pull/36

Thanks! :)

happysalada, 2020-11-11:
I don't use kailh unfortunately. But now I think I see what you mean. The problem I solved with the choc was a different one.

the change in the PR you mentioned is only 14 lines!
https://github.com/tshort/dactyl-keyboard/pull/36/files
I don't have kailh switches to test though, let's leave this open until somebody can test it properly.

c10l, 2020-11-11:
I'm not the OP but I'll be receiving Kailhs some time in the next few days. Happy to test them out and post my results here if someone else doesn't beat me to it. :)

I'd love to open a PR with the changes I linked above or similar, but Clojure is Greek to me and right now I don't have the bandwidth to dig into this unfortunately.

c10l, 2020-11-17:
I've finally received my keyswitches. I've printed one slot of the choc version of this.

It mostly works. I can get the switch into the slot and it's fairly snug but it definitely doesn't hook to the border like it should.

[image]
[image]
[image]

c10l, 2020-11-17:
I've also tried to fit an amoeba PCB there, and indeed like @RaoulDuke-Esq said in the OP, it won't fit due to the depth of the socket in relation to the frame.

It might still work but it won't be too tight a fit and could end up falling off.

[image]
[image]
[image]

happysalada, 2020-11-18:
I've also seen that the amoeba PCB doesn't fit. The choc pcb swap is different, I'll be testing that in the next few days.
I'm receiving a new version of the design with the choc switch. I'll be posting where I am at the moment so that anybody interested in this can have a look at the code for the choc and do something similar for the kailh.

happysalada, 2020-11-23:
[image]

Here is the state of things with the choc switches.
The fit is pretty good. I didn't measure, but it took me 5-10 prints to get it correct.
If anybody is interested in tackling this issue, just look at `use-choc` in the code base. You can probably introduce a case statement for the type of switch used. Clojure might look a little foreign, but you won't need to do anything too complicated. Just modifying number values, printing and trying the result. Of course, feel free to post your question if you have any.

c10l, 2020-12-12:
@RaoulDuke-Esq My merge request https://github.com/ibnuda/dactyl-keyboard/pull/68 has been merged. It adds an option for Kailh switches which seems to work fine for me.

If you could try it out and let us know if it works that would be great!

RaoulDuke-Esq, 2020-12-12:
I checked this out in OpenSCAD and it looks really good!  The slots created will allow the switches to clip in nicely.  Thanks for this!  We should probably close this one and file a separate one for the thickness of the rear wall as it's two separate issues and you've nicely solved the Kailh support.

I'm hesitant to do another print just yet.  There's some things I noticed from the last one that I haven't yet filed issues for (and probably should).  Did you notice any of these with your most recent print?  It looks like they're still there when I preview in OpenSCAD but I would love a sanity check before I drop a bunch of issues: 
- rear wall thickness too much for use of the pro micro holders linked at the bottom of the configuration webpage
- screw inserts off-center on inside wall
- plate does not confirm to geometry of thumb cluster on the outside/bottom wall
- insufficient clearance for keycaps in thumb cluster (caps will hit each other or the walls of the case)

closed 2020-12-12 by RaoulDuke-Esq

c10l, 2020-12-14:
> rear wall thickness too much for use of the pro micro holders linked at the bottom of the configuration webpage

I tried this and I think it's more about the clearance in the hole itself. What happened to me was that the holder pushed the back part of the case too much until it snapped and broke along the printer layer where the top of the hole is.

> plate does not confirm to geometry of thumb cluster on the outside/bottom wall

The generated plate seems to be all kinds of wrong. Once I'm happy with my case I'll create plates that conform to the bottom of the case by myself.

> insufficient clearance for keycaps in thumb cluster (caps will hit each other or the walls of the case)

I haven't noticed that. To be honest I'd rather the keys in the cluster were closer together as the way it is it's quite a reach for me.