ibnuda/dactyl-keyboard #62: Feature: support for 4 keys on last row

cub-uanic, 2020-10-21:
Additionally to "2 keys", "All keys", "No keys" for the last row, it would be fine to have an option "4 keys".
For layouts like 6x5 it would result a key mask like on Kinesis Advantage.
I think to do that the thumb cluster should be moved for one column to the inner side.
Thank you!

ibnuda, 2020-11-07:
something like this? should already been done. on manuform's page, choose "Full" on "Last Row Key Count" and then "Yes" for "Hide Bottom Pinky?".
[image]

closed 2020-11-07 by ibnuda

cub-uanic, 2020-11-08:
I saw that option.
But there are only 3 keys on the bottom row, and I'd like to have 4 keys, one for each finger.
Also, I would like to line-up vertically most inner key on the thumb cluster with inner keys on the main cluster.
The result will look like Kinesis Advantage.

ibnuda, 2020-11-09:
use this then.
```{
  "keys": {
    "columns": 7,
    "rows": 5,
    "thumb-count": "six",
    "last-row": "full",
    "switch-type": "box",
    "inner-column": null,
    "hide-last-pinky": true
  },
  "curve": {
    "alpha": 12,
    "pinky-alpha": 12,
    "beta": 36,
    "centercol": 4,
    "tenting": 15,
    "rotate-x": 180
  },
  "connector": {
    "external": false,
    "trrs": false,
    "micro-usb": false
  },
  "form": {
    "stagger-pinky-y": -13.0,
    "hotswap": false,
    "stagger-index-y": 0.0,
    "wire-post": false,
    "stagger": true,
    "stagger-ring-z": 0.0,
    "stagger-middle-z": -6.5,
    "screw-inserts": false,
    "thumb-offset-x": 6.0,
    "stagger-middle-y": 2.8,
    "height-offset": 4,
    "wide-pinky": true,
    "stagger-index-z": 0.0,
    "thumb-offset-z": 7.0,
    "stagger-ring-y": 0.0,
    "thumb-offset-y": -3.0,
    "stagger-pinky-z": 6.0
  },
  "misc": {
    "keycaps": true,
    "right-side": true,
    "case": true
  }
}
```
[image]