ibnuda/dactyl-keyboard #66: Feature: enable download .clj file

schickhardt, 2020-10-31:
It would be a great option to enable downloading the .clj file for more precise tweaking. To do the main configuration in the generator and then continue manually.

I really miss the feature to change the wall z and xy offset!

Best regards
schickhardt

happysalada, 2020-11-09:
I'm curious what are the wall z and xy offsets. I only looked at the project in its current form.

ibnuda, 2020-11-09:
@happysalada  those two are used for `wall-locate1/2/3` in `common.clj` to determine the inclination of the wall. and those function are, which in turn, used by [`wall-brace` function](https://github.com/ibnuda/dactyl-keyboard/blob/5700671f308209a98319ed2b42c8f608b9989c27/src/dactyl_keyboard/manuform.clj#L749).

@schickhardt, unfortunately, i am unable to do that because i don't know how to make it spits the clojure file. should it read the source code and then transform it ,using some macros, based on the parameters? i'm not sure.