ibnuda/dactyl-keyboard #79: hotswap hangs

wdtamagi, 2021-04-03:
Firstly thank you so much for this project @ibnuda I'm a beginner at this keyboard world but with this generator you helped me a lot!

I'm preparing to print my first dactyl manuform and I made some changes for hold the hotswap socket I made a kind of hangs to  hold in place.

I will test if the socket fit well and then if it works well I can open a PR with this. 

What you think guys?

[image]
[image]
[image]

happysalada, 2021-04-10:
It's a great idea!
I've done the same in a PR for choc style switches.

wdtamagi, 2021-04-10:
Oh nice,

I did it only for mx style hotswap socket...

I tested and I realized we just need a hang in one side of the hotswap socket... so I'm making some changes and adding the led hole as well... let's see how it will be in the printed test.

wdtamagi, 2021-08-24:
Hey  @yejianfengblue,

Sorry for the delay in reply, I'm a little bit busy but I can take a look this week in case you are needing it.

wdtamagi, 2021-08-24:
BTW it worked pretty well the sockets were very tight.

[image]

yejianfengblue, 2021-08-24:
Good. Thank you.

Looooopy, 2021-10-30:
Hi @wdtamagi 

Any news on the hangs that hold the hot swap plug in-place or is it already merged into this repo?
I suppose it is the KAILH hot swap plug you are using?

andyc1234, 2021-12-20:
I would like to build a hot-swappable DM so I am interested in this as well. Without the hangs, do we need to glue the hot-swap sockets in place?