ibnuda/dactyl-keyboard #83: Error when send post request to manuform api

yejianfengblue, 2021-06-26:
The json is copied from https://dactyl.siskam.link/manuform.json

```shell
curl --location --request POST 'https://dactyl.siskam.link/api/manuform' \
--header 'Content-Type: application/json' \
--data-raw '{
    "keys": {
        "columns": 5,
        "rows": 4,
        "thumb-count": "six",
        "last-row": "two",
        "switch-type": "box",
        "inner-column": "normie",
        "hide-last-pinky": false
    },
    "curve": {
        "alpha": 12,
        "pinky-alpha": 12,
        "beta": 36,
        "centercol": 4,
        "tenting": 15
    },
    "connector": {
        "external": false,
        "trrs": false,
        "micro-usb": false
    },
    "form": {
        "hotswap": false,
        "stagger": true,
        "stagger-index-y": 0,
        "stagger-index-z": 0,
        "stagger-middle-y": 2.8,
        "stagger-middle-z": -6.5,
        "stagger-ring-y": 0,
        "stagger-ring-z": 0,
        "stagger-pinky-y": -13,
        "stagger-pinky-z": 6,
        "wide-pinky": false,
        "height-offset": 4,
        "wire-post": false,
        "screw-inserts": false
    },
    "misc": {https://dactyl.siskam.link/manuform.json
        "keycaps": false,
        "right-side": true,
        "case": true
    }
}'
```
```
<html>
<head>
<meta http-equiv="Content-Type" content="text/html;charset=utf-8"/>
<title>Error 500 java.lang.IllegalArgumentException: No matching clause: </title>
</head>
<body><h2>HTTP ERROR 500 java.lang.IllegalArgumentException: No matching clause: </h2>
<table>
<tr><th>URI:</th><td>/api/manuform</td></tr>
<tr><th>STATUS:</th><td>500</td></tr>
<tr><th>MESSAGE:</th><td>java.lang.IllegalArgumentException: No matching clause: </td></tr>
<tr><th>SERVLET:</th><td>-</td></tr>
<tr><th>CAUSED BY:</th><td>java.lang.IllegalArgumentException: No matching clause: </td></tr>
</table>
<pre>java.lang.IllegalArgumentException: No matching clause: 
        at dactyl_keyboard.manuform$columns.invokeStatic(manuform.clj:37)
        at dactyl_keyboard.manuform$columns.invoke(manuform.clj:33)
        at dactyl_keyboard.manuform$key_holes.invokeStatic(manuform.clj:98)
        at dactyl_keyboard.manuform$key_holes.invoke(manuform.clj:79)
        at dactyl_keyboard.manuform$model_right.invokeStatic(manuform.clj:1402)
        at dactyl_keyboard.manuform$model_right.invoke(manuform.clj:1390)
        at dactyl_keyboard.generator$generate_case_dm.invokeStatic(generator.clj:106)
        at dactyl_keyboard.generator$generate_case_dm.invoke(generator.clj:104)
        at dactyl_keyboard.handler$api_generate_manuform.invokeStatic(handler.clj:394)
        at dactyl_keyboard.handler$api_generate_manuform.invoke(handler.clj:322)
        at compojure.response$eval1946$fn__1947.invoke(response.clj:47)
        at compojure.response$eval1868$fn__1869$G__1859__1876.invoke(response.clj:7)
        at compojure.core$wrap_response$fn__3823.invoke(core.clj:158)
        at compojure.core$wrap_route_middleware$fn__3807.invoke(core.clj:128)
        at compojure.core$wrap_route_info$fn__3812.invoke(core.clj:137)
        at compojure.core$wrap_route_matches$fn__3816.invoke(core.clj:146)
        at compojure.core$routing$fn__3831.invoke(core.clj:185)
        at clojure.core$some.invokeStatic(core.clj:2701)
        at clojure.core$some.invoke(core.clj:2692)
        at compojure.core$routing.invokeStatic(core.clj:185)
        at compojure.core$routing.doInvoke(core.clj:182)
        at clojure.lang.RestFn.applyTo(RestFn.java:139)
        at clojure.core$apply.invokeStatic(core.clj:667)
        at clojure.core$apply.invoke(core.clj:660)
        at compojure.core$routes$fn__3835.invoke(core.clj:192)
        at ring.middleware.json$wrap_json_body$fn__7005.invoke(json.clj:58)
        at ring.middleware.json$wrap_json_response$fn__7058.invoke(json.clj:139)
        at ring.middleware.keyword_params$wrap_keyword_params$fn__5597.invoke(keyword_params.clj:36)
        at ring.middleware.params$wrap_params$fn__4063.invoke(params.clj:67)
        at ring.middleware.absolute_redirects$wrap_absolute_redirects$fn__6041.invoke(absolute_redirects.clj:47)
        at ring.middleware.content_type$wrap_content_type$fn__5989.invoke(content_type.clj:34)
        at ring.middleware.default_charset$wrap_default_charset$fn__6013.invoke(default_charset.clj:31)
        at ring.middleware.not_modified$wrap_not_modified$fn__5970.invoke(not_modified.clj:53)
        at clojure.lang.Var.invoke(Var.java:384)
        at ring.adapter.jetty$proxy_handler$fn__472.invoke(jetty.clj:27)
        at ring.adapter.jetty.proxy$org.eclipse.jetty.server.handler.AbstractHandler$ff19274a.handle(Unknown Source)
        at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:127)
        at org.eclipse.jetty.server.Server.handle(Server.java:500)
        at org.eclipse.jetty.server.HttpChannel.lambda$handle$1(HttpChannel.java:386)
        at org.eclipse.jetty.server.HttpChannel.dispatch(HttpChannel.java:562)
        at org.eclipse.jetty.server.HttpChannel.handle(HttpChannel.java:378)
        at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:270)
        at org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:311)
        at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:103)
        at org.eclipse.jetty.io.ChannelEndPoint$2.run(ChannelEndPoint.java:117)
        at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:806)
        at org.eclipse.jetty.util.thread.QueuedThreadPool$Runner.run(QueuedThreadPool.java:938)
        at java.base/java.lang.Thread.run(Thread.java:829)
</pre>
<hr><a href="http://eclipse.org/jetty">Powered by Jetty:// 9.4.22.v20191022</a><hr/>

</body>
</html>
```

yejianfengblue, 2021-06-27:
Seems the api is a bit outdated.

ibnuda, 2021-06-27:
please check it again. https://github.com/ibnuda/dactyl-keyboard/commit/07404c93a0b9ae53959869c47221337466ea7fc3

yejianfengblue, 2021-06-27:
Still error. Is a reboot required? 

```shell
curl --location --request POST 'https://dactyl.siskam.link/api/manuform' \
--header 'Content-Type: application/json' \
--data-raw '{
    "keys": {
        "columns": 6,
        "rows": 5,
        "thumb-count": "six",
        "last-row": "full",
        "switch-type": "box",
        "inner-column": "ergodox",
        "hide-last-pinky": false
    },
    "curve": {
        "alpha": 12,
        "pinky-alpha": 12,
        "beta": 36,
        "centercol": 4,
        "tenting": 15,
        "rotate-x": 180
    },
    "connector": {
        "external": false,
        "type": "rj9",
        "micro-usb": false
    },
    "form": {
        "custom-thumb-cluster": false,
        "height-offset": 4,
        "hotswap": false,
        "screw-inserts": false,
        "stagger": true,
        "stagger-index-y": 0.0,
        "stagger-index-z": 0.0,
        "stagger-middle-y": 2.8,
        "stagger-middle-z": -6.5,
        "stagger-pinky-y": -13.0,
        "stagger-pinky-z": 6.0,
        "stagger-ring-y": 0.0,
        "stagger-ring-z": 0.0,
        "thumb-cluster-offset-x": 6.0,
        "thumb-cluster-offset-y": -3.0,
        "thumb-cluster-offset-z": 7.0,
        "thumb-middle-left-offset-x": -53.0,
        "thumb-middle-left-offset-y": -26.0,
        "thumb-middle-left-offset-z": -12.0,
        "thumb-middle-left-tenting-x": 69.0,
        "thumb-middle-left-tenting-y": 69.0,
        "thumb-middle-left-tenting-z": 69.0,
        "thumb-top-left-offset-x": -35.0,
        "thumb-top-left-offset-y": -16.0,
        "thumb-top-left-offset-z": -2.0,
        "thumb-top-left-tenting-x": 69.0,
        "thumb-top-left-tenting-y": 69.0,
        "thumb-top-left-tenting-z": 69.0,
        "thumb-top-right-offset-x": -15.0,
        "thumb-top-right-offset-y": -10.0,
        "thumb-top-right-offset-z": -5.0,
        "thumb-top-right-tenting-x": 69.0,
        "thumb-top-right-tenting-y": 69.0,
        "thumb-top-right-tenting-z": 69.0,
        "wide-pinky": true,
        "wire-post": false
    },
    "misc": {
        "keycaps": false,
        "right-side": true
    }
[... 67 lines cut ...]
	at ring.middleware.default_charset$wrap_default_charset$fn__6013.invoke(default_charset.clj:31)
	at ring.middleware.not_modified$wrap_not_modified$fn__5970.invoke(not_modified.clj:53)
	at clojure.lang.Var.invoke(Var.java:384)
	at ring.adapter.jetty$proxy_handler$fn__472.invoke(jetty.clj:27)
	at ring.adapter.jetty.proxy$org.eclipse.jetty.server.handler.AbstractHandler$ff19274a.handle(Unknown Source)
	at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:127)
	at org.eclipse.jetty.server.Server.handle(Server.java:500)
	at org.eclipse.jetty.server.HttpChannel.lambda$handle$1(HttpChannel.java:386)
	at org.eclipse.jetty.server.HttpChannel.dispatch(HttpChannel.java:562)
	at org.eclipse.jetty.server.HttpChannel.handle(HttpChannel.java:378)
	at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:270)
	at org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:311)
	at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:103)
	at org.eclipse.jetty.io.ChannelEndPoint$2.run(ChannelEndPoint.java:117)
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:806)
	at org.eclipse.jetty.util.thread.QueuedThreadPool$Runner.run(QueuedThreadPool.java:938)
	at java.base/java.lang.Thread.run(Thread.java:829)
</pre>
	<hr><a href="http://eclipse.org/jetty">Powered by Jetty:// 9.4.22.v20191022</a>
	<hr />

</body>

</html>
```

ibnuda, 2021-06-28:
alright, please use the latest `manuform.json` and do it again. it should be fixed.

https://github.com/ibnuda/dactyl-keyboard/commit/a12904c3cafc2f52aebd0f894926976421b27dc6

yejianfengblue, 2021-06-28:
Work now. Thanks for the quick reply.

closed 2021-06-28 by yejianfengblue