ibnuda/dactyl-keyboard #90: dactyl manuform error when set thumb key count 5 or generate screw inserts

yejianfengblue, 2021-07-19:
@happysalada I see the related code is updated in the wall thickness pull request. Could you please help?
 
## Case 1
Default parameters of manuform page, with thumb key count 5

java.lang.IllegalArgumentException: Don't know how to create ISeq from: dactyl_keyboard.manuform$thumb_post_tl

## Case 2
Default parameters of manuform page, with screw inserts generaed, that is `form.screw-inserts=true`

clojure.lang.ArityException: Wrong number of args (2) passed to: dactyl-keyboard.common/wall-locate3

happysalada, 2021-07-19:
I love those clojure stacktraces, they're crystal clear! :-)
I made a PR https://github.com/ibnuda/dactyl-keyboard/pull/91
However this will probably break the lightcycle.
Just saying this in case it takes time to merge and you want to apply the commit locally.

yejianfengblue, 2021-07-20:
Stacktraces are added.

closed 2021-07-20 by ibnuda

ibnuda, 2021-07-20:
i've put it online and tried it. happysalada's work works.

yejianfengblue, 2021-07-20:
Work too in my test. Thank both for the quick action.