ibnuda/dactyl-keyboard #100: 5 Column Corne with Inserts is broken

khominhvi, 2021-08-27:
* Floating Insert in 1u thumb cluster

[image]

---

* Floating Insert
* Last column still showing

[image]

```json
{"keys":{"columns":5,"rows":4,"thumb-count":"three-mini","last-row":"zero","switch-type":"box","inner-column":"normie","hide-last-pinky":false},"curve":{"alpha":12,"pinky-alpha":12,"beta":36,"centercol":4,"tenting":15,"rotate-x":180},"connector":{"external":true,"type":"trrs","micro-usb":false},"form":{"stagger-pinky-y":-13.0,"thumb-cluster-offset-x":6.0,"hotswap":false,"stagger-index-y":0.0,"thumb-middle-left-tenting-y":69.0,"thumb-top-left-offset-y":-16.0,"wire-post":false,"stagger":true,"stagger-ring-z":0.0,"thumb-top-right-tenting-y":69.0,"wall-thickness":3.0,"stagger-middle-z":-6.5,"screw-inserts":true,"web-thickness":7.0,"thumb-top-left-tenting-z":69.0,"thumb-top-left-tenting-x":69.0,"thumb-middle-left-offset-x":-53.0,"stagger-middle-y":2.8,"thumb-middle-left-offset-z":-12.0,"thumb-middle-left-offset-y":-26.0,"height-offset":4,"thumb-middle-left-tenting-x":69.0,"wide-pinky":false,"thumb-top-right-offset-x":-15.0,"stagger-index-z":0.0,"thumb-top-right-tenting-z":69.0,"thumb-cluster-offset-z":7.0,"custom-thumb-cluster":false,"thumb-top-right-tenting-x":69.0,"stagger-ring-y":0.0,"thumb-middle-left-tenting-z":69.0,"thumb-top-left-tenting-y":69.0,"thumb-top-right-offset-y":-10.0,"thumb-top-right-offset-z":-5.0,"thumb-top-left-offset-x":-35.0,"thumb-cluster-offset-y":-3.0,"thumb-top-left-offset-z":-2.0,"stagger-pinky-z":6.0},"misc":{"keycaps":false,"right-side":true}}
```

dkbarrett, 2022-03-30:
Still experiencing this; Minidox default settings with screw inserts:
[image]

happysalada, 2022-03-31:
Thank you for reporting!
I'm having a look at doing some fixes. I'll update if I get to it.
I didn't want to have the screws have individual offset as I was afraid it might make things too complicated, but that might be the best solution here.

If anyone beats me to a PR, happy to help merge it.