ibnuda/dactyl-keyboard #102: Is there a way to reduce the spacing between keys?

drGarbinsky, 2021-11-22:
I am looking to reduce the space between the keys by around .5 mm  either between rows and or columns. I realize that as the ends of the columns converge based on curvature settings this could cause an issue but I'd like to experiment with such an option. Any idea if there is a way to do that?

yejianfengblue, 2021-11-23:
https://github.com/ibnuda/dactyl-keyboard/blob/84e86a58582954f60f7844ef571f62e69b9fd02d/src/dactyl_keyboard/common.clj#L9-L16

albert-ying, 2022-10-18:
Hi @drGarbinsky, did you manage to do it? The esrille actually provides the small MX keycap that could support a key spacing of 17.5 mm http://www.esrille.com/keyboard/.

I'm wondering if it is possible to create a smaller version of Dactyl

yejianfengblue, 2022-10-19:
> Hi @drGarbinsky, did you manage to do it? The esrille actually provides the small MX keycap that could support a key spacing of 17.5 mm http://www.esrille.com/keyboard/.
> 
> I'm wondering if it is possible to create a smaller version of Dactyl

Another factor that stops you from making a keyboard with less key spacing is the standard key switch size (cherry mx standard width and height is nearly 16mm) and standard keycap size (nearly 18mm).

Besides the cherry mx standard, there is one kind of key switchs called low profile key switch, but they are a bit expensive and less choice. What is worse is there is no compatibility between difference manufacturer. 

See 
- low profile key switch disaster https://youtu.be/2qzxGP9NStc
- low profile key cap with smaller dimension https://mkultra.click/mbk-choc-keycaps, https://mkultra.click/moergo/
- mass production keyboard with low profile key switch and concave surface  https://moergo.com/
- there are many DIY keyboard with low profile key switch, google yourself

albert-ying, 2022-10-20:
Thank you so much @yejianfengblue . The moergo looks SUPER interesting to me!

drGarbinsky, 2022-10-21:
> Hi @drGarbinsky, did you manage to do it? The esrille actually provides the small MX keycap that could support a key spacing of 17.5 mm http://www.esrille.com/keyboard/.
> 
> I'm wondering if it is possible to create a smaller version of Dactyl

I'm working on this now. I'll report back with results.

drGarbinsky, 2022-10-22:
Results 
[image]

drGarbinsky, 2022-10-22:
Results
[image]

albert-ying, 2022-10-24:
It looks nice! How about the horizontal spacing? It seems that there is still some space between columns.

drGarbinsky, 2022-10-24:
That model has a reduced spacing but it is limited by the gap along the top row. Using a flatter profile will allow for a tighter column spacing

drGarbinsky, 2023-03-21:
Here is [the end result](https://www.printables.com/model/314112-dactyl-manuball)
Here is [the forked code](https://github.com/drGarbinsky/dactyl-keyboard)

closed 2023-03-21 by drGarbinsky