ibnuda/dactyl-keyboard #105: Strane L shape added when using TRS jack

BlueWolf220, 2022-01-22:
[image]
this shape was added into the model, and i'm not sure why. how do i remove it?

yejianfengblue, 2022-01-23:
The L hook is atmega32u4 pro micro holder, see https://github.com/yejianfengblue/dactyl-generator-demo#internal-connector-trrs-micro-usb. 

You can manually delete the openscad code by 
1. select openscad memu `window` -> `editor`
2. right click the L hook, there will be a popup, click the `cube`
3. the editor should jump to the code of the L hook
```
union () {
  difference () {
    translate ([-94.9015632882982, 24.40298907331629, 16.208123960789713]) {
      cube ([6, 12, 12], center=true);
    }
    translate ([-95.9015632882982, 23.40298907331629, 16.208123960789713]) {
      cube ([4, 10, 12], center=true);
    }
  }
  translate ([-60.02565784429761, 37.394528789935315, 2]) {
    cube ([19, 12, 4], center=true);
  }
  union () {
    translate ([-73.6256578442976, 37.394528789935315, 2]) {
      cube ([10.2, 12, 4], center=true);
    }
  }
}
```
4. delete difference() and everything inside it
```
  difference () {
    translate ([-94.9015632882982, 24.40298907331629, 16.208123960789713]) {
      cube ([6, 12, 12], center=true);
    }
    translate ([-95.9015632882982, 23.40298907331629, 16.208123960789713]) {
      cube ([4, 10, 12], center=true);
    }
  }
``` 
5. press F5 to refresh. The L hook should be deleted

BlueWolf220, 2022-01-23:
Okay, thank you very much!

closed 2022-01-23 by BlueWolf220