ibnuda/dactyl-keyboard #107: Plate generation is broken

gahag, 2022-03-22:
When trying to open any plate file generated with the attached json, OpenSCAD fails with:
```
ERROR: CGAL error in CGALUtils::applyUnion3D: CGAL ERROR: assertion violation! Expr: G.mark(v1,0)==G.mark(v2,0)&& G.mark(v1,1)==G.mark(v2,1) File: /usr/include/CGAL/Nef_S2/SM_overlayer.h Line: 294 
```

This may be related to #81 , but I'm not able to make it work by changing the number of columns. Please let me know if I can be of any help on debugging this issue, I would really like to build the keyboard.

[json.txt](https://github.com/ibnuda/dactyl-keyboard/files/8328200/json.txt)

yejianfengblue, 2022-03-23:
You issue is not related to issue 81. The detail error message is different. 

I just have basic openscad knowledge and not an expert in this generator. I guess something conflicts. A workaround is delete line 17532 `cube([50, 50, 5], center=true);` or put a star in front of it. But this way will leave a hole in your plate.

[image]

gahag, 2022-03-23:
Thanks a lot for putting your time into debugging this issue! Knowing the problematic line, I can try to work around it and maybe figure out how to fill the hole. Let me know if I can buy you a beer in appreciation :)

yejianfengblue, 2022-03-24:
Find the sponsor link at the [page bottom](https://github.com/yejianfengblue/dactyl-generator-demo#sponsor).

Add below line to the end of scad file to fill the hole 
```
cube ([50, 50, 3], center=true);  // [x, y, z] is the cube size. the 3rd argument 3 is same as the linear_extrude height at line 2.  
```
[image]

[image]

gahag, 2022-03-25:
It worked! Thanks a lot.

closed 2022-03-25 by gahag